enesbcs/rpieasy #272: Ina219 possible to add Gain and ADC settings?

Micha-123, 2023-04-06:


enesbcs, 2023-04-07:
Possible if you specify which options in which forms is needed... selector/textbox, possible values, ranges... as i do not use INA219 since it is implemented i have no idea about its parameters.
Although i can find out myself, if there is a plenty of time...

Micha-123, 2023-04-09:
```
#!/usr/bin/env python
from ina219 import INA219
from ina219 import DeviceRangeError

SHUNT_OHMS = 0.1
MAX_EXPECTED_AMPS = 0.2

def read():
    ina = INA219(SHUNT_OHMS, MAX_EXPECTED_AMPS)
    ina.configure(ina.RANGE_16V, ina.GAIN_2_80MV, ina.ADC_128SAMP)
```

thats my simple script 

SHUNT_OHMS -> Textbox
MAX_EXPECTED_AMPS  -> Textbox

ina.RANGE_16V / ina.Range_32V -> selector of the two possible values 

ina.GAIN_AUTO / ina.GAIN_1_40MV / ina.GAIN_2_80MV / ina.GAIN_4_160MV / ina.GAIN_8_320MV -> selector 5 values

ina.ADC_9BIT / ina.ADC_10BIT / ina.ADC_11BIT / ina.ADC_12BIT / ina.ADC_2SAMP / ina.ADC_4SAMP / ina.ADC_8SAMP / ina.ADC_16SAMP/ ina.ADC_32SAMP / ina.ADC_64SAMP / ina.ADC_128SAMP,  -> Selector with 11 values

enesbcs, 2023-04-12:
Yep addFormFloatNumberBox() only support input with 2 digits... another input may be needed for values smaller than 0.001. This is the first reason i asked for ranges.

enesbcs, 2023-05-29:
Dropped forcing number format, changed to free text input at commit https://github.com/enesbcs/rpieasy/commit/b14193308604bad16edb7f5789e6e30ab628b363

closed 2023-08-06 by enesbcs