enesbcs/rpieasy #274: Compatibility with python 3.11

clumsy-stefan, 2023-05-19:
Since python 3.8 generator based coroutines are deprecated and have been removed in 3.11. After upgrading to the latest OS which includes python 3.11 as default RPIEasy is failing when started.

See also here: https://docs.python.org/3.10/library/asyncio-task.html#generator-based-coroutines

Unfortunately my python knowledge especially about coroutines is not good enough to propose needed changes!

enesbcs, 2023-05-19:
This means replacing the @asyncio.coroutine decorator with async def keyword, and yield from with await.

In other words: i have to go through the integrated Perver webserver line by line and fix it. :D 
Excellent.

Although stable releases like Ubuntu LTS is not yet affected as it is using Python 3.10, this is now only a problem for beta-testers... :) But yes it will be a huge problem in a year when the next LTS arrives, and i guess a lot of Python applications will be affected.

enesbcs, 2023-05-19:
> I tried that and replaced all yield from's and async def... but then I don't get any answer anymore by HTTP (even though the server seems to run)... So it seems to be a bit more than search and replace, sorry for that..

It seems a long term project :D

clumsy-stefan, 2023-05-19:
thanks...

so in the mean time I need to find how I can downgrade to an older python version on my RPI... (I'm running ubuntu/lunar since 2 days ;)

clumsy-stefan, 2023-05-29:
Cool!! Thank you very much!!

I jsut updated it and did a few quick tests... It seems to run fine!!

I'll report if I find any issues!!

Thanks again for the quick fixing!!

clumsy-stefan, 2023-07-02:
Runs for over a month without issues (in fact, I get the feeling that the web interface is more stable)...

So IMHO this issue can be closed!

closed 2023-08-06 by enesbcs