enesbcs/rpieasy #279: Unknown command: PZEMRESET,1

Epik77, 2023-07-26:
Hi, 
I wanted to reset the counter of my pzem004t but the command is not found.
could you fix it?
Thank you

TD-er, 2023-07-26:
The command is `resetenergy` and it seems like the parameter should be the modbus address.

Epik77, 2023-07-26:
same result
11:15:01 : CMD: resetenergy
11:15:01 : Unknown command: resetenergy
or
11:15:51 : CMD: resetenergy,1
11:15:51 : Unknown command: resetenergy,1

TD-er, 2023-07-26:
I already removed my reply, since I overlooked this was about RpiEasy, not ESPEasy ;)

enesbcs, 2023-07-26:
# Available commands:
#  PZEMADDRESS,<currentaddress>,<newaddress>   	- change address of pzem device
#  PZEMRESET,<address>			 	- PZEM reset energy at address

if the PZEM device is not initialized in RPIEAsy or, address is unknown or resetenergy() returns False, then the result is unknown command.

enesbcs, 2023-08-06:
I've added more log output to current release, please upgrade and check console log
https://github.com/enesbcs/rpieasy/commit/a7d1219c2e98045c7ac10afb01f529aed55b88b5

Epik77, 2023-08-11:
Hi this output:

False
15:39:05 : PZEM004 reset error: 'Instrument' object has no attribute '_performCommand'
15:39:05 : PZEM reset command failed!
15:39:05 : PZEM004 reset error: 'Instrument' object has no attribute '_performCommand'
15:39:05 : PZEM reset command failed!
15:39:05 : Unknown command: pzemreset,1

enesbcs, 2023-08-11:
> 15:39:05 : PZEM004 reset error: 'Instrument' object has no attribute '_performCommand'

I see... when i wrote the plugin at 2020 there was a "_performCommand" function. And now it is called as "_perform_command".. magnificent. :D
"Note that the API might change, as this is outside the official API."

Could you please change your
"lib/lib_pzem.py" file both occurences of this:

self.dev._performCommand(66,'')

to these:

self.dev._perform_command(66, b'')

Epik77, 2023-08-12:
Very good
replaced performCommand with perform_command
now the command works correctly:

08:44:52 : Timer 2 started with timeout: 5
08:44:52 : Unit alive: 21
08:44:52 : Unit alive: 13
08:44:54 : CMD: pzemreset,1

[image]

thanks for support :-)

closed 2023-08-12 by Epik77

enesbcs, 2023-08-12:
Thank you for testing, i will add this fix to the next commit!