enesbcs/rpieasy #280: [Rules] some minor bugs/issues

clumsy-stefan, 2023-09-05:
While writing rules i cam e across some smaller issues. One of them I could fix, for the rest my python knoledge is not enough..

Handling of `[var#num]` vs. `[int#num]`:
Normally '[int#num]' should round the variable to the next lower integer whereas '[var#num]' would leave the variable as is. However both of them are treated the same. To correct this I inserted in `commands.py`  before line 1056:
```
        if taskprop[0] == "int":
         tval = int(GlobalVars[sid])
        else:
         tval = GlobalVars[sid]
```
which fixed my problem and `[int#var]` returns an integer now. (btw: I came across this as I calculated a PWM value to set, but using the command `pwm` with a float number fails..)

Calculation of exponentials / powers:
Either I misunderstand or the use of `^` is not working eg. `let,1,2^5` does not work. Also I guess `exp()` or `pow()` is not implemented. Would be an enhancement though...

Rules with `Time#Clock=`:
Not sure about this, but it seems, that wildcards (like `All` for the day or `**:**` for the time) do not work as in ESPEasy. I came across this as I wanted to execute a rule every 5 or 10 minutes (eg: `on Clock#Time=All,**:*0 do`).

enesbcs, 2023-09-10:
> Handling of `[var#num]` vs. `[int#num]`: Normally '[int#num]' should round the variable to the next lower integer whereas '[var#num]' would leave the variable as is. However both of them are treated the same. To correct this I inserted in `commands.py` before line 1056:
> 
> ```
>         if taskprop[0] == "int":
>          tval = int(float(GlobalVars[sid]))
>         else:
>          tval = GlobalVars[sid]
> ```

Sorry, taskprop[0] got its first value at line 1057.
Please add a PR or show at least 10 of the above and below lines also with your code.

> Calculation of exponentials / powers: Either I misunderstand or the use of `^` is not working eg. `let,1,2^5` does not work. Also I guess `exp()` or `pow()` is not implemented. Would be an enhancement though...

RPIEasy is not ESPEasy. It is a python application. I think you may want to use this command, as this is already implemented by the python interpreter:

let,1,2**4

These are the standard Python operators:
https://docs.python.org/3/library/operator.html#mapping-operators-to-functions

Also built any built-in python functions will work, like:

let,1,int(5/2)

https://docs.python.org/3/library/functions.html

> Rules with `Time#Clock=`: Not sure about this, but it seems, that wildcards (like `All` for the day or `**:**` for the time) do not work as in ESPEasy. I came across this as I wanted to execute a rule every 5 or 10 minutes (eg: `on Clock#Time=All,**:*0 do`).

"All" and "**" supported since 2019 as wildcards.
https://github.com/enesbcs/rpieasy/blob/master/commands.py#L1326

Although "*0" approach is not yet supported, you are right. Please add a new issue with details about its usage.

clumsy-stefan, 2023-09-16:
Thanks for the feedback. PR created for the `[int#num]` issue. 

For the power function `**` I'm getting wrong results. The following rules:
```
let,3,[var#2]**2
event,var3,[var#3]
let,2,0
```
Create the log as follows:
```
23:11:02 CMD: let,3,-16.199999999999974**2
23:11:02 CMD: event,var3,-262.43999999999915
23:11:02 CMD: let,2,0
```
It seems to misinterpret the negativ sign... As a workaround the following calculates correctly:
```
let,3,([var#2])**2
``` 

Also I did not find trigonometric or logarithmic functions. But probably I'm looking at the wrong place....

enesbcs, 2023-09-26:
> Thanks for the feedback. PR created for the `[int#num]` issue.

Thanks, merged! 

> Also I did not find trigonometric or logarithmic functions. But probably I'm looking at the wrong place....

Yep, its in the math module. I've imported it in commit https://github.com/enesbcs/rpieasy/commit/1b177d6f247d97e25697bc0461d9d50642114ab9

This is the list of math functions:
['acos', 'acosh', 'asin', 'asinh', 'atan', 'atan2', 'atanh', 'cbrt', 'ceil', 'comb', 'copysign', 'cos', 'cosh', 'degrees', 'dist', 'e', 'erf', 'erfc', 'exp', 'exp2', 'expm1', 'fabs', 'factorial', 'floor', 'fmod', 'frexp', 'fsum', 'gamma', 'gcd', 'hypot', 'inf', 'isclose', 'isfinite', 'isinf', 'isnan', 'isqrt', 'lcm', 'ldexp', 'lgamma', 'log', 'log10', 'log1p', 'log2', 'modf', 'nan', 'nextafter', 'perm', 'pi', 'pow', 'prod', 'radians', 'remainder', 'sin', 'sinh', 'sqrt', 'tan', 'tanh', 'tau', 'trunc', 'ulp']

enesbcs, 2023-12-03:
closed for inactivity

closed 2023-12-03 by enesbcs