enesbcs/rpieasy #282: RPI 4 - Hardware type not shown and unknown GPIO

harrybzh, 2023-12-04:
Good evening,

I got a pretty similar issue as gary got here I think :
https://github.com/enesbcs/rpieasy/issues/201#

With this message: "This hardware has unknown GPIO."

As I got this by running cat /proc/cpuinfo :
```

cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : c03115
Serial          : 100000005bcf396b
Model           : Raspberry Pi 4 Model B Rev 1.5
```

And I can't seem to find c03115 in linux_os.py

Could it be the same issue ?

I have also issues with the autostart with rc.local. It seems the bash script doesn't run properly, but when I call it manually it starts ... any hints on this ?

Thank you

tonhuisman, 2023-12-04:
@harrybzh What is the memory size of your RPi4, 4GB or 8GB? (Doesn't really matter, except the shown info won't be correct if it's the 8GB model 😋)

harrybzh, 2023-12-05:
Just modifying linux_os.py with the new one ... did actually work :) thank you for the reactivity !

PR #283 (2023-12-06): [Hardware] Add support for RPi 4 rev 1.5 hardware
Resolves #282 

Features:
- Add support for RPi 4 rev. 1.5 hardware, both 4GB and 8GB
- Add `.vscode` folder to `.gitignore` file
- Add preliminary support for RPi 5 hardware, both 4GB and 8GB
- Incremented the version number, 1x for RPi4 update, and 1x for RPi5 addition

NB: RPi 5 hardware revisions taken from [official Raspberry Pi documentation](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/revision-codes.adoc)

Suggestion: Use the detection logic shown in that page, including a Pytho [... cut]
closed 2023-12-06 by enesbcs